lovyan03/ESP32_ScreenShotReceiver #2: Multiple screen on one ESP32

a-RN-au-D, 2020-07-10:
Hello, I really love your project and decided to use it as a base for a simple A.R. project I'm working on.
I want to know if it's possible to use the C.S . pin to use two displays on one esp32 instead of using two esp32 with one diplay each ?
I tried defining the CS pin by myself by using  #define TFT_CS  -1 in tft e_SPI and defined the CS pin myself in the codes only output an error because "recv.setup(.." will not work with TFT_CS is undefined. I can post my codes if you want, 
Thanks

lovyan03, 2020-07-10:
Hello, Thank you for your message.
I am not currently using TFT_eSPI for this project. Instead, I am using LovyanGFX.
LovyanGFX is capable of handling multiple panels. First make sure that you can use multiple panels with LovyanGFX.
If you can do that, you will be able to work with multiple panels in this project as well.

a-RN-au-D, 2020-07-12:
Thanks for your quick answer, I have tried to setup your GFX library but couldn't get the 1_simple_use to work with my esp32 board "Lolin esp32 lite". The display I'm using is based on the ST7789 chip, so I have defined "#define LGFX_LOLIN_D32        // LoLin D32 Pro" as mcu but my display stays dark
Here is my pin setup

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS  5  // Chip select control pin
#define TFT_CS_A -1  // Chip select control pin
#define TFT_CS_B -1  // Chip select control pin
#define TFT_DC   2  // Data Command control pin
#define TFT_RST  4  // Reset pin (could connect to RST pin)
#define TFT_BL   32  //

is there a way for me to manually set this in the code?
Thanks

a-RN-au-D, 2020-07-12:
Please ignore my previous post, I was able to run your GFX library by defining  ESP-WROVER-KIT instead of LoLin D32 Pro, 
it was clearly written in your readme too, Sorry I definitively did not catch that, 
**ST7789 (TTGO T-Watch, DSTIKE D-duino-32 XS, ESP-WROVER-KIT)"**
Now I'll try to see it I can run the multi displays like I have in mind, I'll post my codes once I get it working, 
Thanks again

lovyan03, 2020-07-12:
Also see this for ESP-WROVER-KIT. You probably shouldn't choose it.
https://github.com/lovyan03/LovyanGFX/issues/25#issuecomment-657150923

I have added English comments to "examples/HowToUse/2_spi_setting" of the unstable branch. Please take a look at it here.
https://github.com/lovyan03/LovyanGFX/blob/unstable/examples/HowToUse/2_spi_setting/2_spi_setting.ino

And for LovyanGFX questions, please visit the LovyanGFX Issue Board.

a-RN-au-D, 2020-07-14:
I tried configuring my own esp32, but still can't use multiple displays, 
I will close this thread and reopen it in the LovyanGFX if I still can't figure it out, 
Thanks for your help +1

lovyan03, 2020-07-18:
I have updated the "unstable" branches of both LovyanGFX and ESP32_ScreenShotReceiver.
You can use "#include<TCPReceiverDual.h>" to include A single ESP32 can draw a screen for two panels, VSPI and HSPI. 
The usage is " recv.setup( &lcd , &lcdB ); "

Please give it a try.

Dimision, 2020-07-18:
That's really a great project. 
If we use this system for streaming videos, Is it necessary to use UDP protocol instead of TCP protocol to optimise the latency of network transmission?  In my case, I use the 1.3" 240x240 ST7789 panel which has 3-wire SPI bus and works at 40MHz, the serial monitor shows the latency is floating between around 10~300(Quality=60), and I'm not sure it caused by network transmitting or JPG decoding.

lovyan03, 2020-07-18:
Hello, Dimision.
Speeds will be slowed down by WiFi congestion. If there are many access points in the vicinity, the situation will be tougher.
The ESP32 can't handle jumbo frames, so the JPEG data will be in a single packet of UDP. It does not fit. Therefore, packet loss and variation in the order of arrival can cause rendering errors.
This project was created using UDP at first, but due to high packet loss, it could not display a stable picture. The problem with TCP was that it was not able to display a stable picture, so we changed to TCP. I am not going to try UDP as it has been resolved.

Dimision, 2020-07-19:
Hi Lovyan, thanks for your patient replying. 
UDP protocol seems to be a dead end, but I realized the jpgRead function seems try to separate the TCP package into pieces and receive them, I got some ESP32 modules few days ago which integrated 8MB pSRAM, maybe we can use these babies to buffering whole TCP package instead of receiving it piece by piece?

lovyan03, 2020-07-19:
I've already tried it.
You can try. You probably won't get any speed.

By the way, here's an Issue on multi-panels that a-RN-au-D set up.
If you want to continue with more irrelevant replies, please make a new Issue.

lovyan03, 2020-07-20:
@a-RN-au-D
I'm Sorry.
I updated my local branch but didn't update github.
I just added TCPReceiverDual.h to the unstable branch.
Please try it out and see.

lovyan03, 2020-07-20:
@a-RN-au-D 
The WiFiServer class has an initial parameter of "max_clients=4" , so it may be able to maintain a connection with multiple clients, but I don't know if it can communicate with more than one client at a time.
[image]
I think it's a good thing you're trying, but I don't want to complicate the communication part, so I recommend splitting a single JPEG image in two.

a-RN-au-D, 2020-07-20:
Thanks a lot for your quick answer again, I'm not too good at coding but I'm good at finding solutions, 
could you give me some indications on where to look to split the jpgs?

lovyan03, 2020-07-20:
I added "TCPReceiverDual.h" in unstable branch.
Two LCDs are passed to one TCP receiver to make it work.
The usage is " recv.setup( &lcd , &lcdB ); "

https://github.com/lovyan03/ESP32_ScreenShotReceiver/blob/unstable/ScreenShotReceiver/src/TCPReceiverDual.h

a-RN-au-D, 2020-07-20:
That is so awesome!! I Will give it a run now and report the issues I get, 👍 +1

a-RN-au-D, 2020-07-20:
I just did a test to the receiveDual code and this is the output. I believe it doesn't pick up the display size anymore.
here are the pictures: https://imgur.com/a/w8KRdD1, It similar to when the display config isn't correct

a-RN-au-D, 2020-07-20:
Yet another update in my adventure, It worked!! your TCPReceiverDual.h did the job. The reason why it wouldn't work was because I only downloaded only the "TCPReceiverDual.h" files instead of all of them. After doing so, the display worked out of the box. There's is a small issue when passing the screen definition here "  recv.setup( &lcd, &lcdB);", when I switched the panel order in this way " recv.setup( &lcdB,&lcd);", it would render one of the display unusable. 
But now everything works, I'm going to add a gyroscope and see if I can complete my cheap smart glasses,
Thanks again 👍 +1

closed 2020-07-23 by a-RN-au-D