lovyan03/ESP32_ScreenShotReceiver #3: ESP32 and ILI9488

Stke7, 2020-07-14:
I've configured and tested LovyanGFX with the example codes and it displays properly. No issues on the config part but i cant get the display to work with the ScreenShotReceiver. it shows up like this on the screen. 
[image]

lovyan03, 2020-07-14:
Sorry, ScreenShotReceiver is limited to 16-bit color mode to gain frame rate Since the ILI9488 must be sent in 24-bit color over an SPI connection, the The ScreenShotReceiver cannot be displayed properly.

lovyan03, 2020-07-15:
I created an unstable branch with an additional 24-bit mode.
Probably works fine with ILI9488. Please give it a try.

https://github.com/lovyan03/ESP32_ScreenShotReceiver/tree/unstable

Stke7, 2020-07-15:
> 
> 
> I created an unstable branch with an additional 24-bit mode.
> Probably works fine with ILI9488. Please give it a try.
> 
> https://github.com/lovyan03/ESP32_ScreenShotReceiver/tree/unstable

Thanks alot man! love the work you have done. i'll give it a try!

Stke7, 2020-07-15:
Ok i've just tried the unstable branch and it's working! Thank you so very much! I've done some benchmarking for you @60% quality and 40MHz spi write.
7fps at 480x320 no DMA,11fps with DMA
14fps at 320x240 no DMA, 20fps with DMA
50-52fps at 160x120 no DMA, 57-62 with DMA

lovyan03, 2020-07-15:
Congratulations and thank you !!
You've probably set "freq_write=40000000" .
If you can run at the SPI setting of 80000000 you will get double the speed.
I don't know if the ILI9488 will be able to keep up with that speed, and the limit value will vary depending on the wiring conditions.
Try it out if you're interested.

Stke7, 2020-07-15:
I tried it, does not work at 80Mhz. The picture is distorted. Thanks for your Help!

closed 2020-07-15 by Stke7