lovyan03/ESP32_ScreenShotReceiver #6: TCP message type

a-RN-au-D, 2020-08-26:
Hello, I've been trying to quite unsuccessfully to write my own version of the screenShotSender,
I can connect to the esp32, see that it is receiving some bytes, but I can't get the display to turn on. the serial message I get from the device is "Broken data" so that probably means the data isn't right, 

My question is, how should I format the data? I'm sure I'm missing something, could you please direct me on how to format the image to be sent, 
thanks, 

Arnaud

lovyan03, 2020-08-26:
@a-RN-au-D 
hello.
ScreenShotReceiver requires 7 bytes of original data before the jpeg data.
The first three bytes are "JPG"
The following 4 bytes are the length of the jpeg data

https://github.com/lovyan03/ESP32_ScreenShotReceiver/blob/master/ScreenShotSender/FormSenderMain.cs#L229-L236

[image]

a-RN-au-D, 2020-08-26:
Thank you a million time, I'll give it a try right now and hopefully it will work :)

closed 2020-08-31 by a-RN-au-D

a-RN-au-D, 2020-09-05:
Hello it's me again, I just wanted to update you on my project because you were so helpful,
I was able to rewrite a bad version of your screenshotreceiver using processing and get it to send some screenshots, 
I'm using your GFX library for my project so I included you in my boot routine. I hope it's fine with you.
[Before solving the framerate issue](https://www.instagram.com/p/CEuFIeCD7JE/)
[current state](https://www.instagram.com/p/CEwwyhLj9vk/)

I will release everything once I understand github, thanks again for your help 👍

lovyan03, 2020-09-05:
@a-RN-au-D 
Great job!  👍
Thank you for including lovyanGFX's name in your boot routine.
I expect your project to produce something interesting !