lovyan03/ESP32_ScreenShotReceiver #7: ScreenshotSender Support for linux (raspberry pi etc)

matrizxc, 2020-09-22:
Hi, is it possible to use the sender software on Linux instead of windows?

lovyan03, 2020-09-22:
Hello.
I do not develop software on Linux.
You will have to implement it yourself.

Here's what you need to do.
1. connect to ESP32 by TCP
2. wait for "JPG\n" to be sent from ESP32.
3. send the string "JPG" followed by 4 bytes of jpg data size, and then send the actual JPG data.

It looks like there is a project that actually did that.
#6
@a-RN-au-D
https://www.instagram.com/p/CEwwyhLj9vk/

He might be able to give you some advice.

matrizxc, 2020-09-22:
Alright, thank you for your help!

closed 2020-09-22 by matrizxc

a-RN-au-D, 2020-09-26:
Hello matrizxc, I've managed to make send screenshots using processing, my version isn't as polished as Lovyan03's but it works on a mac which is my main platform. my goal is to run this from a raspberry pi but I haven't tested it yet, if you give me another week to polish my codes and learn github, I'll gladly share it with you.

modi12jin, 2022-05-23:
@lovyan03  I found a software that can cast a screen on Android. I have made some attempts on ESP32S3. It may be relatively laggy, but there are places worth optimizing.

https://github.com/dkrivoruchko/ScreenStream

https://user-images.githubusercontent.com/40233017/169747081-f0c69fe7-6d5e-45d9-80a4-8bc524db6bd4.mp4

lovyan03, 2022-05-25:
@modi12jin 
Please don't post on unrelated or closed Issues. I will ignore it.