lovyan03/ESP32_ScreenShotReceiver #10: When using ST7789V parallel port 8 bit in ESP32-S2, there is a color problem

modi12jin, 2021-09-07:
[image]

```
// v1.0.0 を有効にします(v0からの移行期間の特別措置です。これを書かない場合は旧v0系で動作します。)
#define LGFX_USE_V1

#include <esp_wifi.h>

#include "src/TCPReceiver.h"
#include <LovyanGFX.hpp>

class LGFX : public lgfx::LGFX_Device
{

   lgfx::Panel_ST7789 _panel_instance;

   lgfx::Bus_Parallel8 _bus_instance;

   lgfx::Touch_FT5x06  _touch_instance; 

public:
   // コンストラクタを作成し、ここで各種設定を行います。
   // クラス名を変更した場合はコンストラクタも同じ名前を指定してください。
   LGFX(void)
   {
      {                                     // バス制御の設定を行います。
         auto cfg = _bus_instance.config(); // バス設定用の構造体を取得します。

         // 8ビットパラレルバスの設定
         cfg.i2s_port = I2S_NUM_0;  // 使用するI2Sポートを選択 (0 or 1) (ESP32のI2S LCDモードを使用します)
         cfg.freq_write = 20000000; // 送信クロック (最大20MHz, 80MHzを整数で割った値に丸められます)
         cfg.pin_wr = 40;           // WR を接続しているピン番号
         cfg.pin_rd = 42;           // RD を接続しているピン番号
         cfg.pin_rs = 41;           // RS(D/C)を接続しているピン番号
         cfg.pin_d0 = 0;            // D0を接続しているピン番号
         cfg.pin_d1 = 1;            // D1を接続しているピン番号
         cfg.pin_d2 = 2;            // D2を接続しているピン番号
         cfg.pin_d3 = 3;            // D3を接続しているピン番号
         cfg.pin_d4 = 4;            // D4を接続しているピン番号
         cfg.pin_d5 = 5;            // D5を接続しているピン番号
         cfg.pin_d6 = 6;            // D6を接続しているピン番号
         cfg.pin_d7 = 7;            // D7を接続しているピン番号

         _bus_instance.config(cfg);              // 設定値をバスに反映します。
         _panel_instance.setBus(&_bus_instance); // バスをパネルにセットします。
      }

      {                                       // 表示パネル制御の設定を行います。
         auto cfg = _panel_instance.config(); // 表示パネル設定用の構造体を取得します。

         cfg.pin_cs = -1;   // CSが接続されているピン番号   (-1 = disable)
         cfg.pin_rst = -1;  // RSTが接続されているピン番号  (-1 = disable)
         cfg.pin_busy = -1; // BUSYが接続されているピン番号 (-1 = disable)

         // ※ 以下の設定値はパネル毎に一般的な初期値が設定されていますので、不明な項目はコメントアウトして試してみてください。

         cfg.memory_width = 240;   // ドライバICがサポートしている最大の幅
         cfg.memory_height = 320;  // ドライバICがサポートしている最大の高さ
         cfg.panel_width = 240;    // 実際に表示可能な幅
         cfg.panel_height = 320;   // 実際に表示可能な高さ
         cfg.offset_x = 0;         // パネルのX方向オフセット量
         cfg.offset_y = 0;         // パネルのY方向オフセット量
         cfg.offset_rotation = 0;  // 回転方向の値のオフセット 0~7 (4~7は上下反転)
         cfg.dummy_read_pixel = 8; // ピクセル読出し前のダミーリードのビット数
         cfg.dummy_read_bits = 1;  // ピクセル以外のデータ読出し前のダミーリードのビット数
         cfg.readable = true;      // データ読出しが可能な場合 trueに設定
         cfg.invert = false;       // パネルの明暗が反転してしまう場合 trueに設定
         cfg.rgb_order = false;    // パネルの赤と青が入れ替わってしまう場合 trueに設定
         cfg.dlen_16bit = false;   // データ長を16bit単位で送信するパネルの場合 trueに設定
         cfg.bus_shared = true;    // SDカードとバスを共有している場合 trueに設定(drawJpgFile等でバス制御を行います)

         _panel_instance.config(cfg);
      }
    { // タッチスクリーン制御の設定を行います。（必要なければ削除）
      auto cfg = _touch_instance.config();

      cfg.x_min      = 0;    // タッチスクリーンから得られる最小のX値(生の値)
      cfg.x_max      = 239;  // タッチスクリーンから得られる最大のX値(生の値)
      cfg.y_min      = 0;    // タッチスクリーンから得られる最小のY値(生の値)
      cfg.y_max      = 319;  // タッチスクリーンから得られる最大のY値(生の値)
      cfg.pin_int    = -1;   // INTが接続されているピン番号
      cfg.bus_shared = false; // 画面と共通のバスを使用している場合 trueを設定
      cfg.offset_rotation = 0;// 表示とタッチの向きのが一致しない場合の調整 0~7の値で設定

// I2C接続の場合
      cfg.i2c_port = 0;      // 使用するI2Cを選択 (0 or 1)
      cfg.i2c_addr = 0x38;   // I2Cデバイスアドレス番号
      cfg.pin_sda  = 8;     // SDAが接続されているピン番号
      cfg.pin_scl  = 9;     // SCLが接続されているピン番号
      cfg.freq = 400000;     // I2Cクロックを設定

      _touch_instance.config(cfg);
      _panel_instance.setTouch(&_touch_instance);  // タッチスクリーンをパネルにセットします。
    }

      setPanel(&_panel_instance); // 使用するパネルをセットします。
   }
};

static LGFX lcd;
static TCPReceiver recv;

void setup(void)
{
  Serial.begin(115200);
  Serial.flush();

  lcd.begin();
//lcd.setColorDepth(24);
  lcd.setRotation(0);
  if (lcd.width() < lcd.height())
    lcd.setRotation(1);

  lcd.setFont(&fonts::Font2);

  Serial.println("WiFi begin.");
  lcd.println("WiFi begin.");
  // 記憶しているAPへ接続試行
  WiFi.mode(WIFI_MODE_STA);
  WiFi.begin();

  // 接続できるまで10秒待機
  for (int i = 0; WiFi.status() != WL_CONNECTED && i < 100; i++) { delay(100); }

  // 接続できない場合はSmartConfigを起動
  // https://itunes.apple.com/app/id1071176700
  // https://play.google.com/store/apps/details?id=com.cmmakerclub.iot.esptouch
  if (WiFi.status() != WL_CONNECTED) {
    Serial.print("SmartConfig start.");
    lcd.println("SmartConfig start.");
    WiFi.mode(WIFI_MODE_APSTA);
    WiFi.beginSmartConfig();

    while (WiFi.status() != WL_CONNECTED) {
      delay(100);
    }
    WiFi.stopSmartConfig();
    WiFi.mode(WIFI_MODE_STA);
  }

  Serial.println(String("IP:") + WiFi.localIP().toString());
  lcd.println(WiFi.localIP().toString());

  recv.setup( &lcd );
}

void loop(void)
{
  recv.loop();
}
```

 @lovyan03  Hi! How can I solve the color problem?

lovyan03, 2021-09-08:
@modi12jin 
 are you using ESP32-S2?

modi12jin, 2021-09-08:
@lovyan03 
I am using ESP32-S2.
Hope to get support

lovyan03, 2021-09-08:
@modi12jin 
The ESP32-S2 is a single core and has a low clock.
The bottleneck is in JPEG decoding, so using a parallel connection will not give you any speed.
I'm not really interested in supporting it because I know it will be defeated by the ESP32.

lovyan03, 2021-09-08:
@modi12jin 
I just tried it with the ESP32-S2. It was slow, but it worked fine.
The problem of strange colors may be solved by lowering the clock.
ESP32-S3 may eventually support this if it is available.